PiInTheSky/lora-gateway #5: New Feature: Additional metadata logging

g8dhe, 2015-08-03:
Would it be possible to expand the logging info. possibly using an additional config. parameter say;
TelemetryMetadata=<Y/N>
The data to consist of modem config., recieve metadata and raw telemetry.
Channel, Frequency, Imp/Exp, Bandwidth, SF, EC, LDRO on/off, Recieve time, AFC, RSSI, SNR, "raw data"
e.g.
0,434.2950MHz,Explicit,20.8k,SF8,EC4:6,LDRO,201508031408,-0.3Khz,-100,20,"Telemetry"

This would then allow much easier understanding of the performance of a given link after the event.

daveake, 2015-08-03:
Yup, good idea.

g8dhe, 2015-08-11:
OK well I've had ago myself!  You should find the input on the Pull request if I have got the hang of GIT right!

PiInTheSky, 2018-03-26:
Done now, by Phil Crump - see latest (V1.8.15)

closed 2018-03-26 by PiInTheSky