PiInTheSky/lora-gateway #7: RSSI calculated value ?

g8dhe, 2015-08-13:
Dave, I've noticed in the code several places where;
"readRegister(Channel, REG_PACKET_RSSI) - 157)"
but in another place its 
"readRegister(Channel, REG_PACKET_RSSI) - 137)"
The latter corresponds with the manual  section 
"5.5.5. RSSI in LoRaTM Mode
The RSSI values reported by the LoRaTM modem differ from those expressed by the FSK/OOK modem. The following formula shows the method used to interpret the LoRaTM RSSI values.
R SSI[dB m ] = –137 + RSSI"

Is it me or should they all be -137 as opposed to -157 ?
Geoff

daveake, 2015-08-13:
Yeah, they changed it between 2 versions of the manual.  Dunno why I 
have both old and new versions.  I'll sort it.

Dave

g8dhe, 2015-08-13:
Ah, that would make sense, which version is right mines version 1.0 it seems ?

daveake, 2015-08-13:
There's at least one later one.  I'll check tomorrow.

g8dhe, 2015-08-13:
Ah, checking the Semtech SX1279 Rev5 Nov. 2014 datasheet it says the Lowband port (upto 525MHz) should use "RSSI (dBm) = -164 + Rssi, (when using the Low Frequency (LF) port)",
Doh yet another value, although they use -157 for the HF port ..... oh and some notes to go with it as well which suggest it varies with lna settings as well.
Maybe the answer is its relative only!

g8dhe, 2015-08-16:
Latest version is Rev 4 March 2015 from the Semtech site.

daveake, 2016-09-14:
Right, so it's 157 for HF which is band 1 which is 868/915, and it's 164 for bands 2/3 which are 434 and 169.

I'll put it in a function that tests.  This will be in V1.8.3 soon.

And yes, it's more an indication than a calibrated value!

Romeo-Golf, 2016-12-29:
Did this make it into 1.8.3?

PiInTheSky, 2018-02-12:
Yeah, was done a while ago.

closed 2018-02-12 by PiInTheSky