PiInTheSky/lora-gateway #12: How many end-nodes can be connected

drasko, 2015-11-17:
I can not see from here: http://www.hoperf.com/rf/lora/RFM98W.htm which Semtech transceiver this module integrates. I guess it is a custom chip with Semtech IP core.

Anyway, as I understand, for LoRa GW we need:
1. Concatenator - SX1301:
http://www.semtech.com/wireless-rf/rf-t ... rs/sx1301/
2. Transcaiver - SX1276:
http://www.semtech.com/wireless-rf/rf-t ... rs/sx1276/
3. I/Q modulator/demodulator - SX1257:
http://www.semtech.com/wireless-rf/rf-t ... ers/sx1257

How come that this soultion does not need Concatenator and/or I/Q modulator?

How many clinets (end-nodes) can be connected to a GW like this?

fab4space, 2015-11-17:
It integrates a sx1276 transceiver via SPI. The RPI can be connected to 2 SPI devices, so this gateway can handle at most 2 concurrent clients.

PiInTheSky, 2015-11-17:
The gateway uses LoRa (SX1276) but does not use nor is it compatible with LoRaWAN.  It's a specific solution for tracking high altitude balloons which run matching software (https://github.com/piinthesky or https://github.com/daveake/FlexTrack).

PiInTheSky, 2015-11-17:
Re number of clients, the limit is not 2.  Many more can be used with TDMA.

drasko, 2015-11-17:
This TDMA can be implemented in SX1276 FW or in RPi SW, as a LInux app (or driver)?

PiInTheSky, 2015-11-17:
Yes, both clients I linked to support this (AVR/Arduino and C/Linux)

drasko, 2015-11-17:
So - TDMA must be implemented on both client and GW side, and you must have some kind of sync reference...

On the GW side - do you implement TDMA in SX1276 FW or directly in Linux on RPi?

PiInTheSky, 2015-11-17:
Yes, on both sides, using GPS timing.  Every balloon tracker has GPS anyway.

Everything is done as C applications running on the host.

closed 2016-08-19 by PiInTheSky