PiInTheSky/lora-gateway #30: Option to run without ncurses

Romeo-Golf, 2016-12-23:
As running nurses based code and in particular ensuring it starts at boot time, which typically requires the use of 'screen', it would be great if it was possible to have an option to start the gateway in a non-ncurses mode? 

Perhaps albeit at the expense of the interactive options this would be useful to assist with integrating this into other projects... Is this something that could be considered?

Thanks

daveake, 2016-12-23:
It can run at boot time without an actual screen.  A good option is to use
the Linux screen program, so the gateway is attached to that and you can
later connect to the gateway using the screen application.

For integration with other software there is a json feed on a network port.

As a separate project, I'm currently rewriting the gateway as some Python
libraries.

Romeo-Golf, 2016-12-24:
Yeah I had figured that it could be started at boot time but without something like screen then you would have to redirect the output to /dev/null, is there an option to run the gateway script so that it doesn't fire up the nurses interface?

If not then it would be good to have the option to run it as a daemon with all output sent to syslog, but failing that if there was an option to run without curses and have any output simply printed on sequential lines then it would be easy to wrap this up with something like ProcServ... Does the seem reasonable?

Romeo-Golf, 2016-12-24:
Also in terms of integration with the adsb-receiver project at this stage we are primarily looking to provide as many feeder/decoder options as possible to increase its usefulness...

I can certainly see value in integrating flight data for Aircraft (ADSB), Gliders (FLARM) and HAB's...  but both the coding and the agreement between these various projects is probably beyond my capabilities :)

daveake, 2016-12-24:
My LoRa gateway is just one feeder to the balloon position and map, so any
integration should be done using the database (see habhub.org) and not the
gateway.

daveake, 2016-12-25:
Just run it during startup; the ncurses output won't go anywhere.

See the logging options in the readme.

Romeo-Golf, 2016-12-30:
For reference I have got this running in the background via procserv but had to set the TERM value to something other than unknown, i used 'vt220' as this seemed to display as much as i can get from the gateway program until im able to get my LoRa board wired up... but had to set this to 'dumb' in order to suppress the curses based output :)

PiInTheSky, 2016-12-31:
No plans to change this.

closed 2016-12-31 by PiInTheSky