PiInTheSky/lora-gateway #31: Support for Dragino Lora/GPS Raspberry Pi HAT

Romeo-Golf, 2016-12-23:
While it looks like there may be some wiring challenges, it would be useful to confirm if the Dragino Lora/GPS Hat will work out of the box with this project?

http://wiki.dragino.com/index.php?title=Lora/GPS_HAT

Romeo-Golf, 2016-12-23:
And if it doesn't work out of the box it would be useful to document what additional wiring is required based on the current v1.3 board:

https://github.com/dragino/Lora/tree/master/Lora_GPS%20HAT/v1.3

And with any luck it might be possible to have these added to a future revision to take advantage of the convenience offered by having both GPS and Lots radios on a single HAT :)

daveake, 2016-12-23:
It won't work with that, because they've made the mistake of connecting the
NSS LoRa pin to a random GPIO pin instead of one of the 2 standard SPI CS
pins.

Romeo-Golf, 2016-12-24:
Dave,

Thanks KS for confirming, I did suspect that based on the GPS chip's PPS pin not being wires to any gpio pins the wiring of this hat hadn't been well thought out... 

But as I needed a GPS hat and these GPS+LoRa boards were equivalent cost to GPS only I ordered one anyway :)

So looks like I will be using the Black Hat Hack3r board to test wiring the GPS PPS pin into the Pi GPIO, typically on GPIO 17:

https://shop.pimoroni.com/products/black-hat-hacker

Can you advise on the LoRa chip to gpio pin mapping this code is expecting?

Thanks

daveake, 2016-12-24:
You can override the default mapping by using DIO0_0= etc.  See the readme

Romeo-Golf, 2016-12-29:
Ok so the DIO pins can be specified in the config, is it possible to do the same for the NSS pin to say use GPIO pin 6 / physical pin 22 rather than pins 24 and 26 listed in the Readme?

Thanks

daveake, 2016-12-29:
No.  The code as it stands can only use standard SPI CS pins. That's not 
going to change.

Romeo-Golf, 2016-12-30:
Its a shame that the manufactures of this board didnt follow the 'standard', although to be fair there doesnt seem to be an abundance of documentation beyond this:

https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/

I will try physically re-mapping the pins with the hack3r board and if successful then perhaps they would consider more sensible pin mapping in the next revision.

Out of interest is ther DIO pin mapping also based on a standard or reference source?

PiInTheSky, 2016-12-31:
DIOs can connected to any free I/O pin, unlike SPI pins which are fixed.

closed 2016-12-31 by PiInTheSky

pelwell, 2017-01-16:
@daveake FYI, with a suitable Device Tree overlay you can ask the SPI driver to use any random GPIOs as chip selects.

daveake, 2017-01-16:
Cheers, didn't know that.  I'll add some notes to the readme.

pelwell, 2017-01-16:
OK - I'd better write a suitable overlay then. I'll post it here first with instructions, then get it built into standard images.

pelwell, 2017-01-16:
The new overlay is called spi0-cs:
```
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

	fragment@0 {
		target = <&spi0_cs_pins>;
		frag0: __overlay__ {
			brcm,pins = <8 7>;
		};
	};

	fragment@1 {
		target = <&spi0>;
		frag1: __overlay__ {
			cs-gpios = <&gpio 8 1>, <&gpio 7 1>;
			status = "okay";
		};
	};

	__overrides__ {
		cs0_pin  = <&frag0>,"brcm,pins:0",
			   <&frag1>,"cs-gpios:4";
		cs1_pin  = <&frag0>,"brcm,pins:4",
			   <&frag1>,"cs-gpios:16";
	};
};
```
Save the above to spi0-cs-overlay.dts, and compile it using:
```
sudo dtc -@ -I dts -O dtb -o /boot/spi0-cs.dtbo spi0-cs-overlay.dts
```
Then add the following to config.txt (you can change the 12 and 13 or omit either cs?_pin):
```
dtoverlay=spi0-cs,cs0_pin=12,cs1_pin=13
```
The overlay is now in the kernel source tree, and will be included in future builds.

jorgerance, 2017-10-11:
Is this already supported? Just got one of those Dragino boards and I was willing to play with some "Pi in the sky" nodes, but I can't get the tracker to work and the following error appears on screen:

```
Channel 0 power set to FFh
Setting SF=8
Setting Error Coding=5
```

Can anyone please advise? Is it possible to use these Dragino boards as tracking stations?

daveake, 2017-10-11:
The gateway software uses standard SPI CE0/1 pins and the draguino does
not.  This is not going to change.  It may be possible to remap those pins
but this is something you would need to try.

2020rberry, 2018-01-17:
Two questions.  One, would [this](http://wiki.dragino.com/index.php?title=Lora/GPS_Shield) board work with it?  And two, if yes, my board says I'm using the 915MHz range, but when I put that in as the channel it doesn't work.  Any ideas?