PiInTheSky/lora-gateway #33: ambiguous variables

Romeo-Golf, 2016-12-30:
Can you confirm the permitted values for the following variables which have =0 in the gateway-example.txt

implicit_0=0
lowopt_0=0

But show Y/N answers in the README.md

Implicit_<n>=<Y/N>.  e.g. Implicit_0=Y
lowopt_<n>=<Y/N>.  Enables or disables low data rate optimization.

Thanks

daveake, 2016-12-30:
> Implicit_=<Y/N>. e.g. Implicit_0=Y
> lowopt_=<Y/N>. Enables or disables low data rate optimization.

Y=1  N=0

Romeo-Golf, 2016-12-30:
Thanks for clarifying :)

Does the code accept both, or should the documentation and gateway-sample.txt indicate these should be specified in 0/1 format?

PiInTheSky, 2016-12-31:
Any of 1, Y, y, t or T will be accepted to enable a setting; anything else disables it.

closed 2016-12-31 by PiInTheSky