PiInTheSky/lora-gateway #35: Interactive Features

Romeo-Golf, 2017-01-03:
So I have rewired my Lora HAT and it appears to be detected by the gateway binary; however the 'Interactive Features' mentioned in the README do not seem to be functional... 

Is the expectation that, as per the docs, pressing the 'q' key should result in the gateway exiting?

>   LoRa Habitat and SSDV Gateway by M0RPI, M0RJX - V1.8.6                                                                                 
>                                                                                                                                                    
> Channel 0 434.451.0 MHz                                                                                                                           
>  Implicit, 20.80, SF6, EC4:5                                                                                                                       
> 
> 
> 
> AFC                                                                                                                        
> 
> 
>                                                                                                                                                     
> Tracker = 'hostname'
> Channel 0 frequency set to 434.451MHz                                                                                                               
> LoRa Channel 0 DIO0=31 DIO5=26

daveake, 2017-01-03:
Means your DIO5 settig is wrong.

daveake, 2017-01-03:
Check DIO5 setting, that correct SPI CS pin is connected to the LoRa 
module, and that the correct SPI channel is being used.

If the SPI side is correct then the RSSI figure will be correct; 
otherwise it will be locked at -157.

The interactive function does work.

closed 2018-02-12 by PiInTheSky