PiInTheSky/lora-gateway #46: Stop overwriting of SSDV files

hyde00001, 2018-04-23:
In V1.8.19 and possibly earlier, if the SSDV image count wraps from 255 (the maximum SSDV utilises) back to 0, the LoRa Gateway ssdv folder over-writes older images with newer images with the same index count. Could code be added to avoid this, and retain all received images? Perhaps this could be achieved by creating subfolders for each cycle through the SSDV image count on a given day or by prefixing/suffixing the image file name with date and/or time. Obviously, other approaches exist to resolve this behaviour.

Thanks for all the effort maintaining this excellent application.

Steve Hyde - hyde00001

KevWal, 2021-02-28:
I have also found this issue with my flight yesterday, and on a related point the wrap at 255 causes ssdv_resend.py to not work beyond image 255 too. I did have a look through the code but couldn't find anything obvious that was causing the wrap.

Thanks very much
Kevin

daveake, 2021-02-28:
The wrapping is because SSDV uses a single byte for the image number; PITS
and the gateway need to do the same.