PiInTheSky/lora-gateway #56: MaxAFCStep may be used uninitialised

ED6E0F17, 2018-07-10:
I may be missing something, but If I don`t set MaxAFCstep in config, I am not going to be able to use calling mode to correct drifty payloads.

PiInTheSky, 2018-07-10:
If you don't set a value, it will be zero, which disables its action.  So AFC works but isn't bounded.

closed 2018-07-10 by PiInTheSky

daveake, 2018-07-10:
> Thanks, I should have been paying more attention to the code. You could
> set a sensible default value, but it is not a problem.

Yeah, that's reasonable.  I'll add to the list.

Dave

ED6E0F17, 2018-07-10:
My actual problem was a typo in gateway.txt: "AFC0=Y", so if you could make AFC default to on, that would be good too.......

dbrooke, 2018-07-11:
AFC is only needed at the narrower bandwidth settings so it doesn't necessarily make sense to default to on.

Also you can easily toggle the setting using the f or F key.