PiInTheSky/lora-gateway #57: Unable to install dependencies and make

MaartenHeuv, 2018-07-12:
I'm trying to install PiInTheSky on an RPi-II, but I ran in 2 problems.

Trying to install de dependencies I get:
pi@Maarten-pi:~ $ sudo apt-get install git wiringpi libcurl4-openssl-dev libncurses5-dev ssdv
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package ssdv

Ignoring the earlier error I get with the make:
habitat.c:17:23: fatal error: curl/curl.h: No such file or directory
 #include <curl/curl.h>

Thanks for all the works sofar, Maarten

daveake, 2018-07-12:
For SSDV, if you're running Stretch then update it; if earlier then SSDV
wasn't included so install manually:

cd
git clone https://github.com/fsphil/ssdv.git
cd ssdv
sudo make install

Then re-run the apt install line minus the ssdv bit.

Dave

MaartenHeuv, 2018-07-12:
Solved, perfect solution, 

Thanks Maarten

closed 2018-07-12 by MaartenHeuv