PiInTheSky/lora-gateway #59: Uplink Messages - send/receiver using LoRa Expansion Boards?

K4KDR, 2019-05-26:
Hello!  Thank you for all the great work on such an interesting topic.

I recently purchased a couple of the Uputronics Raspberry Pi+ LoRa Expansion Boards (HAB-LORA-4N) to use as receivers in support of a local Amateur Radio Operator / School Teacher who has conducted several successfull HAB flights.

I did not realize that these boards were actually TRANSCEIVERS, not just receivers.  What a nice surprise!

After configuring the proper settings to uplink SMS messages, I verified with an SDR that the devices were absolutely transmitting on the specified frequency.

While I see reference to receiving SMS messages in the PITS repository, is there any way other than purchasing a full tracker unit to receive & decode these uplink messages?  I have two of the LoRa Expansion boards, but when one is configured as receive-only, all it displays when the other unit transmits an uplink message is:  "Unknown packet type is 23h, RSSI -164"

Is it possible to communicate in any usable way between two of these LoRa Expansion Boards since they are in fact transceivers?

Thanks!

-Scott,  K4KDR

daveake, 2019-05-26:
There are some libraries around to transmit raw packets between LoRa 
transceivers, and my suggestion would be that you try one of those then 
you can send and receive whatever data you like.

For Pi In The Sky there are some specific uplink options such as the SMS 
one, and another for asking the tracker to re-send image packets if they 
were missed by the gateway.  Some more will be added and documented for 
(as an example) asking the tracker to release itself from the balloon.

Dave

K4KDR, 2019-05-26:
Thank you so much for the reply!  I'm not a programmer, so things like C+ libraries aren't useful to me but I see that there are some more turnkey solutions along the lines of https://pypi.org/project/pyLoRa/ , so that gives me some hope.  

Even if pyLoRa doesn't support this particular device (will have to investigate), at least there is hope for a Python or similar interface for those like myself that are not equipped to write our own from scratch.  

Links to any known ready-to-use interfaces for these RFM95/98W transceivers appreciated!

daveake, 2019-05-26:
I'd suggest a Python library as that's an easy language to get started with.

K4KDR, 2019-05-26:
Many thanks!!  Yes, I found https://github.com/pmanzoni/pyUputronics and it's working so far - will test further and see what I can do with it.

closed 2021-04-20 by PiInTheSky