PiInTheSky/lora-gateway #65: XL1278-SMT 100% CPU usage

matburnham, 2021-02-21:
I ended up with a batch of  XL1278-SMT modules so figured I'd try and protoboard a LoRa gateway.

[image]

After a bit of faffing with pin numbers, it all seems to work initially, but then it seems to hang (no interactive keystrokes work except Ctrl-C; 100% CPU usage). My additional debug info shows it's correctly reading the SX1278 module version, etc. It seems to go wrong when it starts trying to receive.

[image]

[image]

```
 $ sudo gdb
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) attach 680
Attaching to process 680
[New LWP 685]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
0x0002530c in startReceiving ()
(gdb) info threads
  Id   Target Id                             Frame
* 1    Thread 0x75f15520 (LWP 680) "gateway" 0x0002530c in startReceiving ()
  2    Thread 0x75f14230 (LWP 685) "gateway" __GI___poll (timeout=-1, nfds=1,
    fds=0x75f13c10) at ../sysdeps/unix/sysv/linux/poll.c:29
(gdb) bt
#0  0x0002530c in startReceiving ()
#1  0x0002780c in setupRFM98 ()
#2  0x00012320 in main ()
(gdb) thread 2
[Switching to thread 2 (Thread 0x75f14230 (LWP 685))]
#0  __GI___poll (timeout=-1, nfds=1, fds=0x75f13c10)
    at ../sysdeps/unix/sysv/linux/poll.c:29
29      ../sysdeps/unix/sysv/linux/poll.c: No such file or directory.
(gdb) bt
#0  __GI___poll (timeout=-1, nfds=1, fds=0x75f13c10)
    at ../sysdeps/unix/sysv/linux/poll.c:29
#1  __GI___poll (fds=0x75f13c10, nfds=1, timeout=-1)
    at ../sysdeps/unix/sysv/linux/poll.c:26
#2  0x76e62690 in waitForInterrupt () from /lib/libwiringPi.so
#3  0x76e62730 in interruptHandler () from /lib/libwiringPi.so
#4  0x76d40494 in start_thread (arg=0x75f14230) at pthread_create.c:486
#5  0x76cc3578 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

I'm not great gdb user, but I think it's stuck waiting for an interrupt. I can manually short DIO0 to ground to get it out of this state, but that doesn't really fix the problem.

Has anyone else tried to use a XL1278-SMT module rather than the RFM98 or DRF1278F, etc.? It's got an XS1278 on board, so it should 'just work'. DIO5 is an edge connector rather than through-hole so there's no chance of using that without a breakout.

[image]

It's quite possibly my wiring given I'm dead-bugging 0.05" pitch, but the fact I'm able to read at least the version register gives me hope.

matburnham, 2021-02-22:
[image]

Tonight I lashed up the only DRF1278F I've got on a breadboard, and was finding the same issues initially, but once I lashed up DIO5 and added it to the config it sprang into life:

[image]

Remove the DIO5 connection and set `DIO5_0=-1` and it all hangs up again.

Digging about, I note 7d501594c27eb10449a6a77c502234e564f9125c removed the check some while ago. Hmmm. Does that also mean DIO5_0=-1 it not supported any more? If so, it could do with the README.md reflecting that; and possibly a warning in the code if it's set to that! If not, where do I look next? What exactly do DIO0 and DIO5 do and what are they used for?

daveake, 2021-02-22:
I did mention this before:

DIO0 is "packet received"
DIO5 is "mode changed"

There is currently no provision in the gateway to ignore DIO5 (I was
thinking of PITS which does).  If you don't have DIO5 then you need to
comment out the code that uses it, and uncomment the following delay line,
all in gateway.c:

        while ( digitalRead( Config.LoRaDevices[Channel].DIO5 ) == 0 )
        {
        }
        // delay(1);

Then recompile/link with make

Dave

PiInTheSky, 2021-02-23:
Please update to V1.8.42 which disables use of DIO5 if set to -1

Dave

matburnham, 2021-02-23:
Excellent. Thanks @daveake. That threw me off a little. Once I've got a breakout able to get DIO5 from the XL1278-SMT I'll add DIO5 back in, but in the meantime that'll enable me to have something to receive my tracker on.

closed 2021-02-23 by matburnham

daveake, 2021-02-23:
I wouldn't bother; a 1ms delay is irrelevant.