PiInTheSky/lora-gateway #70: Lots of: Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'

KevWal, 2022-09-04:
Hi

I have just upgraded (by moving the old install out the way and doing a fresh install), copied over my gateway.txt, added the "EnableSondehub=Y" entry and I am getting lots of errors:

```
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
21:10:46 Ch0: $$KW01,277,20:10:44,52.32273,-0.70620,00093,0,0,5,21.7,3.898,0.66,0.00000,0.00000,0.0,0,-62,6,2,N*B44C
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry' 
21:10:47 Ch0: SSDV Packet, Callsign KW01, Image 38, Packet 46
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
21:10:49 Ch0: $$KW01,278,20:10:47,52.32273,-0.70621,00092,0,0,5,21.8,3.898,0.66,0.00000,0.00000,0.0,0,-62,6,2,N*7226
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
21:10:50 Ch0: SSDV Packet, Callsign KW01, Image 38, Packet 47
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry' 
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
21:10:52 Ch0: $$KW01,279,20:10:50,52.32275,-0.70622,00092,0,0,5,21.8,3.898,0.66,0.00000,0.00000,0.0,0,-62,6,2,N*9E92
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'

```

My full gateway.txt is:

```
##### Your details #####
tracker=G7PMO_PiRx
Latitude=52.322
Longitude=-0.706
Radio=LoRa
Antenna=Colinear

##### Config Options #####
EnableHabitat=Y
EnableSondehub=Y
EnableSSDV=Y
JPGFolder=ssdv
LogTelemetry=Y
LogPackets=Y
CallingTimeout=120
ServerPort=6004
#SMSFolder=./
EnableDev=N

##### Config CE0 #####
#frequency_0=434.650
#frequency_0=434.712
frequency_0=434.718
mode_0=1
AFC_0=N
DIO0_0=3
DIO5_0=26

#bandwidth_0=125K
#implicit_0=0
#coding_0=5
#sf_0=8
#lowopt_0=0
# Power PA_MAX_UK = 0x88 = 136
# Power PA_MAX_BOOST = 0x8F = 143
power_0=136

UplinkTime_0=2
UplinkCycle_0=60
UplinkFrequency_0=434.737
UplinkMode_0=6
SSDVUplink_0=Y

## Callingmode is 433.650 mode 5

##### Config CE1 #####
#frequency_1=869.850
#frequency_1=433.650
frequency_1=434.708
mode_1=1
AFC_1=N
DIO0_1=6
DIO5_1=25

#bandwidth_1=125K
#implicit_1=0
#coding_1=5
#sf_1=8
#lowopt_1=0
#power_1=136

#UplinkTime_1=5
#UplinkCycle_1=60
```

Any help appretiated.

Thanks very much
Kevin

KevWal, 2022-09-04:
Restarting the gateway, once the payload had a valid GPS, and the errors go away:

```
21:40:06 Ch0: $$KW01,847,20:40:05,52.32275,-0.70617,00091,0,0,7,21.3,3.887,0.66,0.00000,0.00000,0.0,0,-65,6,3,N*AFA9                
21:40:08 Ch0: SSDV Packet, Callsign KW01, Image 45, Packet 6                                                                        
21:40:09 Ch0: $$KW01,848,20:40:07,52.32275,-0.70617,00090,0,0,7,21.2,3.887,0.66,0.00000,0.00000,0.0,0,-65,6,3,N*752D                
21:40:10 Ch0: SSDV Packet, Callsign KW01, Image 45, Packet 7
```

I wonder if it is pre-gps lock packets trying to be resent?

Just loosing GPS lock doesnt break it:

```
21:45:09 Ch0: SSDV Packet, Callsign KW01, Image 46, Packet 14                                                                       
21:45:10 Ch0: $$KW01,945,20:45:09,52.32275,-0.70615,00089,0,0,0,22.4,3.920,0.66,0.00000,0.00000,0.0,0,-65,6,3,N*C106                
21:45:12 Ch0: SSDV Packet, Callsign KW01, Image 46, Packet 15                                                                       
21:45:13 Ch0: $$KW01,946,20:45:11,52.32275,-0.70615,00089,0,0,0,22.8,3.920,0.66,0.00000,0.00000,0.0,0,-65,6,3,N*C95D 
```  
But restarting the payload does break it:

```
21:47:28 Ch0: $$KW01,1,00:00:00,0.00000,0.00000,00000,0,0,0,27.1,4.381,0.66,0.00000,0.00000,0.0,0,0,0,0,N*B896                      
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
21:47:30 Ch0: Null uplink packet                                                                                                    
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
21:47:31 Ch0: $$KW01,2,00:00:00,0.00000,0.00000,00000,0,0,0,27.1,4.381,0.66,0.00000,0.00000,0.0,0,0,0,0,N*8A97                      
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
21:47:32 Ch0: Null uplink packet                                                                                                    
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
21:47:34 Ch0: $$KW01,3,00:00:00,0.00000,0.00000,00000,0,0,0,27.1,4.381,0.66,0.00000,0.00000,0.0,0,0,0,0,N*6B77                      
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'           
```                           

But this time getting a valid GPS signal did fix the errors:

```
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
21:51:29 Ch0: $$KW01,87,20:51:30,0.00000,0.00000,00000,0,0,3,21.6,3.946,0.66,0.00000,0.00000,0.0,0,0,0,0,N*8A2C                     
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'                                                
[... cut ...]
21:51:34 Ch0: SSDV Packet, Callsign KW01, Image 48, Packet 85                                                                       
21:51:35 Ch0: $$KW01,89,20:51:35,52.32273,-0.70620,00066,0,0,4,21.6,3.946,0.66,0.00000,0.00000,0.0,0,0,0,0,N*E155                   
21:51:36 Ch0: SSDV Packet, Callsign KW01, Image 48, Packet 86                                                                       
21:51:38 Ch0: $$KW01,90,20:51:38,52.32273,-0.70619,00067,0,0,4,21.5,3.946,0.66,0.00000,0.00000,0.0,0,0,0,0,N*C63B
```

Cheers
Kev

daveake, 2022-09-04:
OK I'll check for valid GPS.  Surprising that that gets rejected.

PiInTheSky, 2022-09-05:
Fixed in V1.9.1.

closed 2022-09-05 by PiInTheSky