PiInTheSky/lora-gateway #71: Unexpected HTTP response 400 for URL 'https://api.v2.sondehub.org/amateur/telemetry'

RoelKroes, 2022-09-16:
Hello,

In v1.9.1 I get, after some time of correctly uploading to Sondehub Amateur, a HTTP response 400. This error keeps rolling over the screen.

So basically, the software is uploading just fine and then after some time I get this error and I need to restart as the error keeps on scrolling over the screen.

Not sure if it is me, the software or sondehub amateur.

See screenshot: https://drive.google.com/file/d/1OuMVPzjV9Tyjx-lXQRKAFpnRxq_NH2oU/view?usp=sharing

Thanks,

Roel.

daveake, 2022-09-16:
In the telemetry it looks like the altitude is zero.  Does the 400 only
happen when the altitude is zero?

I'm going to stop it from retrying on a 400 since retries make no
difference.  It would be good to find out why the server is reporting 400
from what looks like good telemetry.

Dave

RoelKroes, 2022-09-17:
I tested this and I can confirm that the error only shows when altitude is zero. Both negative and postive values for altitude give no problems.

Roel.

PiInTheSky, 2022-09-17:
Thanks.  I'll take this up with the sondehub team.  I'll also have the
gateway not retry after a 400.

Dave

darksidelemm, 2022-09-17:
To confirm what's going on here, we really need an example of the JSON data that is being uploaded. 

We do have a check on the lat/lon/alt values, but the check is just that they are a number (either a float or an int), and it *shouldn't* throw an error on the value being 0.

PiInTheSky, 2022-09-18:
Fixed in V1.9.2 which is released now.  See the readme for the changes.

closed 2022-09-18 by PiInTheSky