PiInTheSky/lora-gateway #74: Gateway Pi showing received data in gibberish

jaygould, 2023-03-29:
Hi all, I am sending data from a RFM98 (attached to an ESP32) to the other RFM98 attached to a Raspberry Pi 4b running LoRa gateway. The sending module is using [arduino-lora](https://github.com/sandeepmistry/arduino-LoRa).

The Pi is receiving the data but it's showing an incorrectly formatted string which looks like it's not been decoded correctly:

[image]

Not sure what the issue could be. Here's my sending code for reference:

```
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);
  LoRa.setPins(4, 2, 15);
  LoRa.setSignalBandwidth(125E3);
  LoRa.setCodingRate4(7);
  LoRa.setSpreadingFactor(7);
  LoRa.enableCrc();

  if (!LoRa.begin(434E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  LoRa.beginPacket();
  LoRa.print("$$hadie,");
  LoRa.print(counter);
  LoRa.print(",10:42:10,53.786244,-2.290401,27799.3,1:10\n");
  LoRa.endPacket();

  counter++;

  delay(4000);
}
```

PiInTheSky, 2023-03-29:
I'm not familiar with that library.  The HAB gateway doesn't do any
decoding; it simply shows the string that the LoRa module has received.

You could try different settings for LoRa; I suggest just setting mode 1 in
the gateway and matching those settings at the transmitter.

Other than that I have no suggestions.

Dave

jaygould, 2023-03-29:
@daveake I've been banging my head against the wall since yesterday afternoon and all it took was to change the mode from `0` to `1`. Thank you so much! :)

daveake, 2023-03-29:
Excellent!

closed 2023-08-24 by PiInTheSky