PiInTheSky/lora-gateway #75: Sondehub HTTP 202 error

OZimmermann, 2023-08-20:
Hi, after starting the gateway, the following error appears after some time 
 "Unexpected HTTP response 202 for URL 'https://api.v2.sondehub.org/amateur/telemetry'"

But error.txt file contains error 400
Example: 21:20:22: 400 response to: [{"software_name": "LoRa Gateway","software_version": "V1.10.0","uploader_callsign": "TrEsLora868","time_received": "2023-08-18T19:20:22Z","payload_callsign": "TrEs2Lora868","datetime":"2023-08-18T21:19:22Z","lat": 51.24919,"lon": 6.18008,"alt": 48,"frequency": 868.6625,"modulation": "LoRa Mode 1","snr": 7,"rssi": -55,"raw": "$$TrEs2Lora868,173,21:19:22,51.24919,6.18008,00048,12,40.6,0.0,0.000,0123456A9P*70F8","sats":12,"temp":40.6,"batt":0.0,"uploader_position": [ 51.366, 6.413, 0],"uploader_antenna": "868_Self_Construction"}]

The gateway has normal internet, because the photos are transferred correctly to my ftp-server
and ping api.v2.sondehub.org also works correctly.

I attach errors.txt telemetry.txt packets.txt and gateway.txt. 
Also attach screenshot showing gateway with error 202 and ping to api.

[errors.txt](https://github.com/PiInTheSky/lora-gateway/files/12388757/errors.txt)
[packets.txt](https://github.com/PiInTheSky/lora-gateway/files/12388758/packets.txt)
[telemetry.txt](https://github.com/PiInTheSky/lora-gateway/files/12388759/telemetry.txt)
[gateway.txt](https://github.com/PiInTheSky/lora-gateway/files/12388760/gateway.txt)
[image]
[image]
[image]

Want to start a new flight in two weeks on 2023/09/01 :-) 
Oliver

daveake, 2023-08-22:
Thanks for the detailed report.

202 is caused if the uploaded time is too far in the future, and from the logs I can see that the upload time is far ahead of the actual UTC time from the tracker.

All times should be UTC.  The gateway gets the UTC time from the operating system.

So, please check your system clock settings on your Pi receiver.  You might have the current time and/or timezone set wrongly, so when UTC is calculated from those the result is incorrect.

PiInTheSky, 2023-08-22:
Please see https://linuxize.com/post/how-to-set-or-change-timezone-in-linux/

Dave

PiInTheSky, 2023-08-24:
Closed as believed to be an error in Pi system time / timezone.  Tests with correct settings show that the upload works as expected.

closed 2023-08-24 by PiInTheSky